PalouseRobosub/robosub-2018 #1: Investigate ROS and Forking

irwineffect, 2016-09-10:
The vision system currently forks a bunch of processes for doing individual vision processing tasks. Some research needs to be done to determine if this could cause any issues with the ROS framework, and develop a suitable workaround if necessary.

Floating-Imp, 2016-09-17:
This article may be helpful:
http://wiki.ros.org/rosspawn

Currently it is a proof of concept implementation intended to be a smaller more narrow use case of roslaunch. The article explains this much better.

irwineffect, 2016-09-19:
I'm not sure if rosspawn solves the particular problem presented we have here. We want to use fork so that we don't need to redo current vision code. But yeah rosspawn may work as a substitute if necessary.

irwineffect, 2016-09-19:
I did some messing around with forking, it looks like everything plays nicely together, no issues with duplicate nodes or topic messages getting rerouted to the child process. Converting the vision module therefore should be straightforward. Take a look at the fork_parent and fork_child nodes in the example folder: https://github.com/PalouseRobosub/robosub/tree/392de7ec3e7263b10c488102561a9426e9ac6eb3/src/examples

closed 2016-09-19 by irwineffect