PalouseRobosub/robosub-2018 #3: Investigate Dynamic Parameters

irwineffect, 2016-09-10:
ROS has a framework for dynamically reloading parameters, which would be useful for several nodes on the sub, including tuning PID parameters. This task is to learn how dynamic parameters work and to create an example node demonstrating how to use them. 

Starting point: http://wiki.ros.org/dynamic_reconfigure/Tutorials

irwineffect, 2016-09-19:
So, I just discovered this: http://wiki.ros.org/roscpp/Overview/Parameter%20Server#Cached_Parameters

This basically is exactly what I wanted. Normally, getting a value from the parameter server takes about 2 ms, using the cached version of the call takes about 33us. I think we can close out this issue.

skallaher, 2016-09-20:
@irwineffect, I guess I didn't really clear up what your use case was for these features. I don't understand quite all of the ins and outs of Cached Parameters, but from what I gathered asking @bkallaher, they can't dump to a file which I would think to be useful with pool testing. If you could clarify what you were thinking of using these for, that would be great.

irwineffect, 2016-09-22:
kk sounds good

closed 2016-09-22 by irwineffect