PalouseRobosub/robosub-2018 #8: Write Sensor Calibration Service

irwineffect, 2016-09-10:


irwineffect, 2016-09-15:
db458f5a61c61a8be7624b0d30bc605373db2075

skallaher, 2016-10-19:
I am in need of a build test of the BNO arduino code with the added changes to verify that it works as well as some code review on the BNO function calls.

Using "rosservice call /bno_trim_service" should execute the added changes and zero out the BNO.

irwineffect, 2016-10-19:
We're currently working on integrating the compilation of Arduino code, I'm not a fan of the arduino-cmake system, looking into integrating with platformio.

skallaher, 2016-12-09:
@irwineffect, I believe this was made obsolete by #87, is this correct?

irwineffect, 2016-12-12:
@ryan-summers, can we add this feature to our bno055 node?

ryan-summers, 2016-12-12:
Will add in future. Feel free to assign to me.

ryan-summers, 2016-12-12:
General issue though is that I don't believe quaternion work through subtraction like Euler does.

irwineffect, 2016-12-12:
Does the BNO have anything build in?

ryan-summers, 2016-12-13:
A quaternion is an orientation matrix, so according to the following post they can be added and subtracting by multiplying together (addition) or multiplying by the inverse (subtraction)
Aka:
```
Quaternion offsets;
...
Quaternion reading;
Quaternion normalized_reading = reading * offsets.inverse();
```

ryan-summers, 2016-12-13:
And to answer your question, the BNO does not have anything builtin that I can see.

skallaher, 2017-03-08:
Completed by #210.

closed 2017-03-08 by skallaher