PalouseRobosub/robosub-2018 #9: Write UPS ROS Node

irwineffect, 2016-09-10:


bkallaher, 2016-09-11:
Might consider using the BatteryState message: http://docs.ros.org/api/sensor_msgs/html/msg/BatteryState.html

irwineffect, 2016-09-12:
Agreed. Issue description updated accordingly

Floating-Imp, 2016-09-13:
I was thinking about adding in some features similar to what already exists in upsc, upssched and upsmon with ROS integration. I've made an incomplete UML diagram of what I'm sort of thinking.
[image]

@irwineffect, any thoughts? The general idea is that the node would be able to not only signal other nodes to shutdown before shutting down the computer, but it would be able to run other types of commands that might be useful to different nodes based upon event occurrences. There are some details that I would need to work out, but my main question is: Is it worth doing and if so is there anything you would add or remove?

irwineffect, 2016-09-13:
I'll admit I don't really know how to read UML diagrams, let's talk about it in the lab.

skallaher, 2016-09-23:
For future reference, these are useful links:

[NUT ups.status possible values](http://networkupstools.org/docs/developer-guide.chunked/ar01s04.html)
[NUT variable list](http://networkupstools.org/docs/user-manual.chunked/apcs01.html)

closed 2016-09-23 by irwineffect