PalouseRobosub/robosub-2018 #47: Integrate microcontroller firmware into build system

irwineffect, 2016-10-19:
We want to be able to compile the microcontroller code when we compile all our other software. Ideally, it should also be simple to upload the software after compilation if desired.

Possible methods:
[rosserial_arduino cmake integration](http://wiki.ros.org/rosserial_arduino/Tutorials/CMake)
[platformio](http://platformio.org/)
[custom use of arduino cmake](http://www.hygy.hu/Tech-blog_m69/Using-arduino-from-command-line-in-Ubuntu-with-cmake_c1)

skallaher, 2016-12-09:
@irwineffect, is this necessary anymore after #87?

irwineffect, 2016-12-12:
Yes, we probably will have other microcontroller code for LEDs, pneumatics, depth, etc.

ryan-summers, 2018-01-31:
PlatformIO has been selected as the defacto tool. Closing for now unless further discussion is warranted.

closed 2018-01-31 by ryan-summers