PalouseRobosub/robosub-2018 #72: Rework UPS Node

irwineffect, 2016-11-10:
Make the UPS publish more stuffs!

bkallaher, 2016-12-11:
@irwineffect I currently have added temperature, runtime, input voltage, and input current. I haven't tested to see if we get all of these through the NUT driver yet. Anything else seem useful? See the changes in commit 771b075.

bkallaher, 2016-12-12:
Available Variables from the UPS:
battery.capacity: 100
battery.charge: 50
battery.charge.low: 5
battery.charge.warning: 20
battery.current: 0.000
battery.mfr.date: ?
battery.runtime: 3932100
battery.temperature: 31.22
battery.type: ?
battery.voltage: 0.01
device.mfr: Mini-Box.Com
device.model: OPEN-UPS
device.serial: PBSO4-LiFePO
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.version: 2.7.1
driver.version.data: openUPS HID 0.1
driver.version.internal: 0.38
input.current: 0.000
input.voltage: 17.87
output.current: 0.500
output.voltage: 12.01
ups.mfr: Mini-Box.Com
ups.model: OPEN-UPS
ups.productid: d004
ups.serial: PBSO4-LiFePO
ups.status: OL
ups.vendorid: 04d8

irwineffect, 2016-12-12:
Looks good! Idk if we need to publish the battery chemistry or the name, those are both constant values that I'm not sure if we really care about. So for the voltage variable, is that the current voltage of the battery, or the output voltage going to the computer? Same for the current variable.

bkallaher, 2016-12-12:
The voltage and current are of the battery. If you want we can also get the output voltage and current. I'll also rename the voltage and current variables to something like voltageBattery and currentBattery.

bkallaher, 2016-12-12:
Also can you think of any special things that the main battery might give us that we would want to know (Once the new board is done)?

bkallaher, 2016-12-12:
I'm also going to add a pretty printer soon.

irwineffect, 2016-12-12:
The main battery also should give us individual cell voltages.

bkallaher, 2016-12-12:
I'll add in an array for cell voltages to the message.

bkallaher, 2016-12-13:
Check out my latest push and let me know what you think.

irwineffect, 2016-12-13:
looks good!

bkallaher, 2016-12-15:
Completed by #113

closed 2016-12-15 by bkallaher