PalouseRobosub/robosub-2018 #119: Control System Relative 0 Bug

irwineffect, 2016-12-18:
An issue that has been observed is that when repeatedly sending a relative depth 0 command to the control system the sub will slowly float to the surface, instead of the expected behavior of maintaining the current depth.

The reason this is happening is that when using purely proportional control on the control system, there is always steady-state error on the depth (e.g. if you request 1 meter, you will actually be at .9ish meters depending on the proportional value). Setting a relative 0 command will set the new absolute goal to be the current state, which is .9 meters, at which point the error decreases causing the sub to float up to the new steady state error, which becomes the new goal.

The current logic is roughly:
```
if new_goal_type is RELATIVE:
    current_goal = current_state = new_goal
```

To fix this problem, I propose having a special case for updating the control system state when getting a relative goal:
```
if new_goal_type is RELATIVE:
    if new_goal is 0:
        if current_goal_type is ABSOLUTE:
            current_goal = current_goal
        else:
            current_goal = current state
     else:
         current_goal = current_state += new_goal
```
This problem was first observed by @skallaher, and confirmed by myself.

bkallaher, 2016-12-18:
Which case will happen more often 0 or not 0? Just wondering if the logic should be more of:
```python
if new_goal_type is RELATIVE:
    if new_goal is not 0:
        current_goal = current_state += new_goal
    else:
        if current_goal_type is ABSOLUTE:
            current_goal = current_goal
        else:
            current_goal = current state```

skallaher, 2016-12-20:
@irwineffect, I am waiting on this to do further (more accurate) testing of the AI. Speaking of which, we need to decide on how we want to structure the AI. There are three options that I can see, two of which I have example branches for.

1. Use roslaunch to launch up the given node with remap arguments and the like (see `sean/aiWrapper` branch).
2. Using actionlib to send and receive messages to already running task nodes (see `sean/actionAI` branch). This has the downside of, from what I can tell, being strictly C++ with how I have it implemented as well as having a lot of nodes constantly running.
3. Using a nodelet system (perhaps in conjunction with actionlib) to squeeze all of the AI nodes underneath a nodelet manager.

I am open to any suggestions at this point.

irwineffect, 2016-12-24:
@skallaher, let's move this AI discussion to #125

irwineffect, 2016-12-24:
@bkallaher, I've got no idea which will be more common. Will we be on target or off target more often? It's a different answer for each axis, depth is more likely to be correct than say, yaw. In any case, I don't think we really need to worry about optimizing this right now.

bkallaher, 2016-12-24:
It was just something to think about.

skallaher, 2016-12-26:
Fixed in #127.

closed 2016-12-26 by skallaher