PalouseRobosub/robosub-2018 #121: Control launch file references nonexistent files

bkallaher, 2016-12-20:
The launch file for the control system references a thruster.yaml and control.yaml that don't exist. Update the control launch file to use the cobalt.yaml file that exists.

skallaher, 2016-12-20:
Is the include folder within launch also outdated? It contains `movement_ironman.xml` and `sensor_ironman.xml`. If so, the movement launch file is outdated, though something should be in place to specify which thrusters to use.

skallaher, 2016-12-20:
Also, the `sim.launch` file references the unity simulator. This seems like it should also be removed along with any other references to the outdated simulator. Thoughts @BigDCross?

BigDCross, 2016-12-20:
Yea kill sim.launch.

PR #124 (2016-12-24): Updated control.launch to use cobalt.yaml
The control launch file now uses the cobalt parameter file or any other sub param file given that the argument `sub_name` is set correctly (defaults to "cobalt"). Also removed sim.launch as it references the old simulator. This fixes #121.
closed 2016-12-24 by irwineffect