PalouseRobosub/robosub-2018 #145: Documentation!!!!!

BigDCross, 2017-01-15:
- [ ] Vision
- [ ] Controllers
- [ ] Localization
- [ ] Start-up and Running Software
- [ ] High Level Node Descriptions
- [ ] Launch File Descriptions
- [ ] Parameter Descriptions
- [ ] Rqt Tips and Tricks
- [ ] Github and Pull Review Processes
- [ ] Senior Design Documentation
- [ ] AI
- [ ] Getting Started Page Rework
- [ ] ROS Command Line Guides

bkallaher, 2017-01-15:
I done some of the CLI guides. I've done the followng so far:
- roslaunch
- rosconsole
- rosed
- rosls
- roscp
- roscd
- roscat
- rosbag

bkallaher, 2017-01-15:
I have also updated the controllers page to include instructions on running the nodes for the various controllers.

skallaher, 2017-01-15:
I believe that there are a few pages that are most definitely outdated at this point and should probably be removed. 

e.g. Communcation System, RoboVision (Unless we want to resurrect it), Tools (Though other pages could probably be moved underneath it such as the YAML Formatting Script page.) 

The socat page should also be rethought as it is just a single line of text with a command.

I would also like to motion for a namespacing of the pages. Many of them are connected, but they are all just under the Computer Science space. For instance, the Cameras, Camera Calibration, and Vision pages could all go under one Vision section. Doing this would clean up the look of the page tree.