PalouseRobosub/robosub-2018 #163: Update Thruster Driver class

BigDCross, 2017-01-22:
Change the MaestroController class to be the MaestroDriver class.
Update the Maestro class to take in a linearized data from 0-100% force. This class should then take this value and map it to the proper value as specified in the T200 documentation. It should account for deadband and non-linear properties of the thruster.

ryan-summers, 2017-01-23:
I'll take this. I'm preparing the bluerobotics graphs for regression currently.

ryan-summers, 2017-01-23:
[image]
Here's a plot of the cubic exponentials I used for modelling the thruster curves. The key was to ensure that the curve was fit for negative data and positive data separately so that the deadband didn't screw with the fitting.

ryan-summers, 2017-01-23:
For posterity, I am going to be adding this data and the figures, as well as the python script used to generate the polynomials.
[curve_fitting.zip](https://github.com/PalouseRobosub/robosub/files/724258/curve_fitting.zip)

ryan-summers, 2017-01-23:
Note that the above curve_fitting.zip file contains an incorrect assumption in the data. Please use the following curve_fitting.zip
[curve_fitting.zip](https://github.com/PalouseRobosub/robosub/files/724635/curve_fitting.zip)
The corrected graph is as follows
[image]

irwineffect, 2017-02-08:
fixed by #167

closed 2017-02-08 by irwineffect