PalouseRobosub/robosub-2018 #213: Write Control Message Helper Functions

ryan-summers, 2017-03-06:
Currently, the control message is pretty opaque and we frequently malform messages. This is largely a result of the enums being fairly unintuitive to a human. Write some C++/BASH/python scripts for modifying a control message. For example:
```
set_depth(message, depth);
add_depth(message, depth);
hold_depth(message);

set_pitch_down(message, pitch_down);
add_pitch_down(message, pitch_down)
hold_pitch_down(message);
//etc.
  ```
Where set_x would set the message to ABSOLUTE value, add_x would set RELATIVE value, and hold would set RELATIVE 0.

bkallaher, 2017-07-04:
I have the basics for this complete (see [bkallaher/pycontrol](/palouserobosub/robosub/tree/bkallaher/pycontrol)). The main question now is who should be doing the broadcasting of the message, my class or the AI?

bkallaher, 2017-07-04:
Do we also want there to be checks for valid inputs inside this class?

irwineffect, 2017-07-06:
I'm thinking your constructor should take in a publisher object that the AI initializes, and then you can have a `publish()` method that just internally calls `self.pub.publish(self._control_msg)`

What sort of valid input checking are you thinking of? My gut reaction is nah, I wouldn't worry about. We can just keep it simple.

irwineffect, 2017-07-06:
Oh, also I understand why you put it the class in `src/movement/`, but you need to think about how an AI module is going to import it. ROS can setup your PYTHONPATH so you can just say `import robosub.src.movement.control_wrapper` or something like that, but you need to configure the CMakeLists.txt and maybe the package.xml.

bkallaher, 2017-07-06:
If you want the publishing to be done by the wrapper would it be better if the wrapper creates the publisher from a passed in topic name? Would you prefer that the class be moved into either the scripts or ai folder? As for input checks I was thinking mostly for the absolute goals that the passed in values are in range (-180 to 180 for rotation and -50 to 50 for linear).

irwineffect, 2017-07-07:
Actually yeah good idea on creating the publisher yourself with passed in topic name. You can have that be defaulted to `control` so we don't have to worry about it unless we want to override it. Let's just put it in the AI folder for now. Hmm I still think it's fine to not have checks, no reason to over complicate this class for now.

bkallaher, 2017-07-27:
Fixed by #265

closed 2017-07-27 by bkallaher