PalouseRobosub/robosub-2018 #221: Cleanup Camera Drivers

skallaher, 2017-03-19:
This could end up being an in-depth process and would most likely require adding the drivers to the robosub Debian repository.
- [ ] Investigate modifying the pointgrey camera drivers to remove many of the unnecessary topics published to.
- [ ] Investigate modifying the pointgrey camera drivers to publish standard Image messages rather than WFOV Images.

This would also include general cleanup in the [camera launch file](https://github.com/PalouseRobosub/robosub/blob/dev/launch/cameras.launch).

- [ ] Remove unnecessary independent processing (Lines 21-25 and 42-46)
- [ ] Add bottom facing camera and allow for it to not be present

irwineffect, 2017-03-20:
For reference, the drive code is here: 
https://github.com/ros-drivers/pointgrey_camera_driver

skallaher, 2017-04-14:
This is now High Priority as testing camera calibration and stereo vision needs this to be completed.

PR #246 (2017-05-12): Camera Driver Fix
Requires #245 commits. (These will be removed from this PR after #245 is merged into dev)

Updates the camera launch file to remove unnecessary processing nodes and switches undistortion system to using `image_raw` which is of message type Image rather than the old WFOVImage on `image`.

This fixes #221.

---
This change is [[image]](https://reviewable.io/reviews/palouserobosub/robosub/246)
closed 2017-05-12 by irwineffect