PalouseRobosub/robosub-2018 #228: Make general purpose math header.

BigDCross, 2017-03-28:


skallaher, 2017-03-28:
This would entail a creating a header file in the [`utility` directory](https://github.com/PalouseRobosub/robosub/tree/dev/src/utility) containing any mathematical operations used repeatedly in systems such as the [control system](https://github.com/PalouseRobosub/robosub/tree/dev/src/movement) and localization system.

irwineffect, 2017-04-05:
Let's also add an Euler distance function, for both 2 and 3 dimensions:
```c++
double euler_distance(double x, double y);
double euler_distance(double x, double y, double z);
```

lucy9tran, 2017-04-10:
@bkallaher Where is the simulation branch/folder? I think you said there were math functions needed for the math header there too?

skallaher, 2017-04-10:
@lucy9tran, the simulator is a separate repository [here] (https://github.com/PalouseRobosub/robosub_simulator). Look at the simulator bridge in particular. Don't worry about making the simulator reference your code, we can have @BigDCross do so when you're merged into dev. On another note, you should talk with @irwineffect about having a cop file for your actual source code and how to modify the CMakeLists file.