PalouseRobosub/robosub-2018 #234: Pneumatic Arduino Code

irwineffect, 2017-04-07:
## Introduction
This is going to be an arduino-based node that actuates our pneumatics to fire torpedoes and drop markers. It should use the ROS services API. Need to talk to the mechanical engineers to ask them how to actuate the pneumatics to move the mechanisms.

## Service Names
`fire_torpedo/left`
`fire_torpedo/right`
`drop_marker`

## Parameters
Exact parameter naming is tbd, but they will be describing which of the pnematics perform which action. For example pn1 might be used to open the marker dropper, and pn5 might be used to reset the marker dropper mechanism.

skallaher, 2017-04-09:
Since there are two marker dropper mechanisms, shouldn't there be two services? One for left and one for right?

It also might be a good idea to have a way to determine the status of each so that the AI doesn't have to guess if it has already fired a torpedo. This could be a service call as well.

irwineffect, 2017-04-10:
Agreed, updated the description!

bkallaher, 2017-04-10:
see #236 for the addition of the tbd service.