PalouseRobosub/robosub-2018 #237: LED Arduino Code

irwineffect, 2017-04-10:
## Introduction
This is going to be an arduino-based node that controls our LEDs. It should allow for both precise control over the pixels and switching between pre-defined patterns.

## Topic Names
`led/raw` - raw control over the pixels, should be an array of rgb values
`led/solid` - sets all pixels in the array to the color specified
`led/pattern` - sets the pixels to some predefined pattern

## Parameters
`led/count` - the number of pixels in the array

irwineffect, 2017-04-10:
Note the above API is still under construction, people are welcome to comment and refine the API.

skallaher, 2017-04-11:
What about having these be services? Do we want to implement the system where each subsystem has a section of the led's that display its status while running? Perhaps allow preset patterns similar to the old system?

I suggest a parameter for the refresh FPS for the LEDs.

How are the LEDs going to be arranged on the sub and should that information be used within the arduino node?

irwineffect, 2017-04-11:
These are all fantastic questions. I was just thinking of this issue as getting the basic functionality working at first, to get someone new familiar with how to work the LEDs. Yeah picking a pattern or color might be worth it as a service call, but I think the raw control should be a topic for speed.

ryan-summers, 2017-05-08:
Update. The LEDs are currently only monochronic, and you can only set them percentage on and off.