PalouseRobosub/robosub-2018 #248: Suspected Memory Leak

irwineffect, 2017-05-01:
I logged into my computer today and discovered that I was completely out of RAM and swap after running the simulator for a few days. Not sure which node(s) is causing the leak, more investigation needs to be done. I first suspect the vision processing, I noticed a lot of ram was freed when I killed the buoy nodes.

Running nodes:
- robosub_simulator/gazebo.launch
- robosub/vision.launch
- robosub/jirwin_gate_task.py

Sindorman, 2017-05-13:
Did you run simulator for days without stopping?

skallaher, 2017-05-13:
@Sindorman, that seems to be the case. Have you seen any symptoms of this problem?

Sindorman, 2017-05-14:
@skallaher I tried to recreate same. I run simulator making sub face buoys, I also was running rqt plagin of left camera(if that helps). I took a screenshot of some of htop information, and left it run over night. I did notice change of total memory usage from 2.56 GB to 2.86-3.06 Gb, however the percentage of memory usage stayed the same on simulator. But the number of threads were 20 more then when I left it.

irwineffect, 2017-10-25:
Have not been able to reproduce, closing for now.

closed 2017-10-25 by irwineffect