PalouseRobosub/robosub-2018 #258: IMU Architecture Overhaul

irwineffect, 2017-07-04:
We need to properly add support for multi-IMU. There are a few things we need to figure out when implementing this:

## Node Naming
If we have multiple of the same sensor, how do we name them? For example, (`bno055_0`, `bno055_1`) or (`bno055_left`, `bno055_right`)?

## Topics
IMUs should not publish to any global topics (such as /orientation or /pretty/orientation). Instead, they should publish within their own namespaces, for example: `/bno055_left/orientation` assuming the node is named `bno055_left`.

We will use a generic IMU node that we can configure to subscribe to one or more topics to process and rebroadcast the orientation data to /orientation and /pretty/orientation. If we only have one IMU, we should use a [topic relay](http://wiki.ros.org/topic_tools/relay).

## Parameters
How do we do parameters so that nodes load them up properly? For example, we currently use /ports/<sensor_name> to load up the port. Should we continue doing this? How does a bno055 node know to load from `/ports/bno055_left` or  `/ports/bno055_right`?

I think this API may prove helpful, we can namespace the nodes to get their topics under their own name, and they can lookup their name to figure out which port parameters to load.
http://wiki.ros.org/roscpp/Overview/Names%20and%20Node%20Information

skallaher, 2017-07-04:
@irwineffect You haven't mentioned anything about fusion.

irwineffect, 2017-07-04:
Oh, that's what I meant by "process and rebroadcast" when I was talking about the generic IMU node.

skallaher, 2017-07-06:
I prefer the explicit naming rather than indexing as it makes it easier to quickly identify the correct hardware. I most definitely agree with modifying the parameter names. "sensor" is too ambiguous. You should be able to accomplish the output topic results rather easily by making the nodehandle it is published through a private nodehandle.

skallaher, 2017-07-13:
Are any parts of this unimplemented by #262?

irwineffect, 2017-07-13:
Nah, this was fixed by #262

closed 2017-07-13 by irwineffect