PalouseRobosub/robosub-2018 #260: Parameter Reorganization

irwineffect, 2017-07-06:
Currently we group parameters like so:
```yaml
ports:
    bno055_left: "/dev/bno055_left"
    bno055_right: "/dev/bno055_right"
    trax: "/dev/trax"
rate:
    bno055_left: 20
    bno055_right: 20
    trax: 20
```

While it makes the parameter file easier to read, it makes code for loading parameters a little more complex, especially when we want to move nodes to different namespaces. I think we should reorganize it like so:

```yaml                                                                 
bno055_left:                                                                    
    port: "/dev/bno055_left"                                                    
    rate: 20                                                                    
                                                                                
bno055_right:                                                                   
    port: "/dev/bno055_right"                                                   
    rate: 20                                                                    
                                                                                
trax:                                                                           
    port: "/dev/trax"                                                           
    rate: 20 
```

This makes putting nodes in different namespaces WAY easier.

skallaher, 2017-07-07:
I remember that there was a reason we decided to do this, but it might be obsolete by now. I like the way you are suggesting.

irwineffect, 2017-07-08:
Lol I stated the original reason in the issue, it makes the parameter file easier to read.

irwineffect, 2017-07-08:
This will be partially fixed by #262, however there are still instances of this in the codebase that need to be fixed.

skallaher, 2017-07-08:
I would also be in favor of splitting up the parameter file into easier to manage chunks. For instance, a `sensors.yaml` and `movement.yaml` where the former would contain the imu, depth sensor, and hydrophone parameters and the latter would contain the thruster and control system parameters. `cobalt.yaml` has gotten pretty big.

irwineffect, 2017-07-08:
Yeah I'm still on the fence about that. I agree that `cobalt.yaml` have gotten large.

The main disadvantage I can think of is that working with multiple param files is more effort than just one. I kinda like that I only need to load one file, I don't need to think about which param file contains the parameters my node needs. With a single settings file, there is zero confusion about which file we should put parameters in, and also eliminates the potential issue of having the same parameter defined in multiple files.

My question is, other than working with smaller files, what do we gain by breaking it apart?

skallaher, 2017-07-08:
Looking forward at reorganizing launch files, it can help prevent loading parameters twice. If there is a `sensors.launch` file it would load the `sensors.yaml` file and help prevent collisions.

irwineffect, 2017-07-09:
That's only true though as long as we don't make the mistake I noted above. If we do accidentally put the same parameter in multiple files we'll then have the strange behavior of different values being loaded for a parameter depending on load order, which is an easy fix, but only after you figure out the cause. Whereas if we have one launch file, it doesn't matter if we load multiple times because it's just overwriting the same values. 

However, if you want the launch file to be capable of overwriting parameter values (I'm not aware if we have a use case for this at this point in time),  then you're right, my system would cause collisions. There is a work around for that though, we just don't have any launch file load the parameter file, we just have to load it manually. Alternatively we could say only the uppermost launch file can load the parameter file. It only needs to be done once after roscore is started.

I guess I don't see loading parameters twice (with the same value) as being a major issue. Have you experienced this being a problem for you?

skallaher, 2017-07-10:
What about instances where only part of the system is desired to launch? For instance, the sensors only without the control system. There would need to be a separate launch file just for sensors. If only the top level launch file loaded param​s, the user would need to remember to do so manually. In the case of having every launch file load the parameters, if one launch file is launched and then later another file is launched, we risk overwriting any parameters that might have been modified during the runtime. Let me know if any of these are non-issues. I can probably explain it better in person.

irwineffect, 2017-07-10:
Yeah, in my mind it is ok for a launch file to crash because you forgot to load the parameter file because it'll be really obvious. The issues I'm concerned about are silent failures. But yeah you're right if we have any nodes that set parameters during run time. I don't think any nodes currently do this though. Do any of the vision nodes do it?

skallaher, 2017-07-10:
Not internally, no. Though if someone were to be messing with the parameters during run-time and they get reloaded, it could lead to a lot of frustration. A lot of the other nodes have their parameters in separate files, such as the localization, ai, and vision. `cobalt.yaml` almost feels like the odd one out.

ryan-summers, 2018-02-03:
The current discussion of breaking up the cobalt parameter file is best served for another issue (I'm in favor of keeping it as is, now that we have purged some of the params - James makes a good point about having zero confusion over the param files). Additionally, I believe having multiple param files may just cause confusion when launching sub software. Separate param files shoud be used typically for private params for a node (such as the BNO configurations, but they're deprecated now).

closed 2018-02-03 by ryan-summers