PalouseRobosub/robosub-2018 #263: Remove delayed_node.sh

irwineffect, 2017-07-07:
`delayed_node.sh` is deprecated in favor of `sleepme.sh`. Replace all usages and remove `delayed_node.sh`.

skallaher, 2017-07-07:
Is this fixed by #261?

irwineffect, 2017-07-08:
No, #261 introduced `sleepme.sh`. This issue is to remove all current usages of `delayed_node.sh`.

skallaher, 2017-07-08:
The only place `delayed_node.sh` should be used is in launch files. None of the launch files in this repo currently delay nodes at all. The `gazebo.launch` file in the simulator repo does. This seems like it would be an issue better suited there.

irwineffect, 2017-07-08:
Good point. Issue is now moved to PalouseRobosub/robosub_simulator#96

irwineffect, 2017-07-08:
Actually this is incorrect, we use `delayed_node.sh` in several tests, so this issue is still valid for this repo.

PR #264 (2017-07-12): Removed instances of delayed_node.sh
Fixes #263 on this repo.

---
This change is [[image]](https://reviewable.io/reviews/palouserobosub/robosub/264)
closed 2017-07-12 by irwineffect