PalouseRobosub/robosub-2018 #301: Creating launch files for pool tests and competition

brennanpeter, 2017-09-24:
We need to create some easy to use launch files for both pool testing and competition. Hopefully we can have an easy to use system for launching the sub when testing but in order to do this we also need to decide what we want launched most of the time

skallaher, 2017-10-01:
A few notes on things to change/mess with:

#### Needed:

1. Update [`vision.launch`](https://github.com/PalouseRobosub/robosub/blob/master/launch/vision.launch). It is outdated as it was intended to be used with the old OpenCV system.
2. Update [`ai.launch`](https://github.com/PalouseRobosub/robosub/blob/master/launch/ai.launch) with new master state machine (This is still in development).
3. Write a `movement.launch` file which starts the thrusters and control system with appropriate parameters.
4. Update parameter files to reflect new structure and remove outdated vision param files.

#### To Discuss/Think About:

1. Have vision.launch start up on the Jetson automatically if on the sub.
2. Include the `ai.launch` in `core.launch`. This will make sense only if the AI's first state waits for the start switch.
3. Look at clean ways to monitor nodes that are launched through `core.launch` when debugging on different machines.
4. Only launching the cameras when on the sub.

Sindorman, 2017-11-15:
To continue @skallaher idea 2. all AI files have implemented `start_switch` as first state.

ryan-summers, 2018-02-03:
I believe we have had a slow progression in launch system development. After the vision purge, sensors launch update, and undistortion updates, a unified cobalt.launch file is proposed in https://github.com/PalouseRobosub/robosub/pull/380 that merges in vision launch (cameras + undistortion), movement (control + thrusters), and parameter loading.  None of these nodes are typically stopped and restarted, so this is a stable basis for launching the submarine software.