PalouseRobosub/robosub-2018 #311: Define localization test parameters

connor-wool, 2017-10-13:


connor-wool, 2017-10-13:
We need to come up with an idea of what our testing suite will look like. We'll probably have to have Gazeebo running so that the hydrophone data and IMU data is being read into the sub. Then, we can have the sub export it's localization info, and compare it to the known location data that we're reading off of the gazeebo engine.

skallaher, 2017-10-13:
@connor-wool, please put your above comment as the description for the issue so it is easier to reference.

skallaher, 2017-10-17:
@connor-wool, what kinds of metrics are you planning on using to evaluate the localization? You should know what your tests are testing before defining how to test them.

connor-wool, 2017-10-19:
For the first series of tests, I think that we should simply compare the data being pushed out by /real/position (in Gazeebo) to the data being pushed out by /position.

The hardest part will be determining how "real world" the sensors in our virtualization system are to the real sub. Especially considering the 180 degree drift @ryan-summers saw on the IMU this last weekend. 

For now, we'll worry about making sure the virtualized system runs properly, and then we'll worry about the uncertainty of our physical measuring devices once we get to the point that we're running this in pool tests.

connor-wool, 2017-10-19:
I propose we use the "sum squared" measurement (using data pulled from /real/position and /position) and run initial tests using the keyboard controls. 

One of my hopes is that we can find some sequence of movements that helps the sub "recalibrate" the localization if we see it fall below a certain threshold of certainty. Perhaps we can point the sub directly along the vector of the hydrophone reading and do a quick forward->back?

connor-wool, 2017-10-19:
Just based on quick qualitative testing (viewing the rostopic echo for /real/position and /position) the best method of recalibrating the position is just to keep the sub stationary for a period of 10-15 seconds. All 3 position fields fall within 0.5 of their expected values after about 15 seconds of reading data.

I expect this is because the kalman filter now is not feeding back into the system, (velocity and acceleration both zero) so the point cloud can begin to converge on a single location.

btmoore, 2017-10-19:
At the moment, if you want the sub to get the best idea of its position, you would have it hold still while maximizing the differential depth between sub and pinger. The smaller the differential depth, the worse your range-estimate will be.

When you have your filters working properly and communicating between each other properly, then your best localization results should come from having the sub:
-continually sampling pings 
-while moving at a constant speed 
-in a direction _orthogonal_ to the source of the pings 
-while retaining significant differential depth. 

The more of these things you have the submarine do, the better its localization will be.  But in practical terms, that is relying on accurate estimate of speed via thruster/drag kinematic models and accelerometer integration, and accurate heading based off magnetometer readings.  Until you have that, you're better off remaining stationary to get your position.

connor-wool, 2017-10-19:
Yeah, that was something that I found out tonight. I was doing some tests, and usually the Kalman filter feeds back into the particle filter. I removed that feedback loop to see the behavior of the particle filter by itself, and it actually was really bad.

TL;DR the code for the actual localization is really good, pretty much complete pending testing and bug-fixes.

I say we switch to testing mode from here on out, and move to tweaking parameters like update time , size of the particle cloud, and the methods used for randomizing the particle field in between processing steps.

btmoore, 2017-10-19:
Does the particle filter have its own update step, where it shifts all the particles based on estimated velocity?  If the Kalman Filter was responsible for substituting that step, then disabling the KF may not be a fair test - it would only prove that he feedback the KF provides is crucial, not that it's working as perfectly as you'd like it to.

That said, if you think your Kalman-Particle filter hybrid is working 'properly', then the 'best' conditions for the subs determining its position should be the following:

-Moving at a constant velocity 
-In a direction _orthogonal_ to the bearing of the pinger
-while maintaining a large differential depth relative to the pinger.

Localizing while moving in a straight line would actually be a pretty good test.  If it performs _better_ than remaining stationary (on average across many trials), that'd be a good indication the system is working as you'd want it to.  Same goes with comparing the accuracy for being stationary at different depths.  Make conditions more and less ideal for the sub, and make sure your localizer performs appropriately under the different circumstances.

closed 2018-07-30 by connor-wool