PalouseRobosub/robosub-2018 #312: Check that data is posted as proper ROS types

connor-wool, 2017-10-13:


connor-wool, 2017-10-13:
As Sean mentioned today, some of the data is being posted in a format other the clean ROS datatypes that we want to be pushing. We should be able to fix this, and it's better to do it sooner so that we can have a standard API that other groups can subscribe to.

connor-wool, 2017-10-13:
Referencing the photos in the Slack chat: Looks like we want to be publishing a single ROS "point", and we might be pushing out a Vector3 right now. 

If someone is really attached to Vector3, we can stick with that (It's essentially equivalent to a point) but I think that for outside use, having the data published as a "point" makes it clearer what the intention of the data is. A vector might confuse someone into thinking that we're pushing out velocity or acceleration data.

skallaher, 2017-10-13:
@connor-wool, You are correct, a Point is more representative of the meaning behind the data.

skallaher, 2017-10-17:
Also a note on this subject, there's no harm in having publishers solely for debug/visualization output as if no one subscribes to it, I believe that there is no overhead involved. This also falls along the lines of premature optimization which should be avoided wherever possible. In other words, there is no such thing as outputting too much data.

connor-wool, 2017-10-19:
Ok, for now, we will focus on trying to add an additional "point" subscription, instead of re-writing the existing. This allows us to non-destructively debug the new output type, and once it's polished we can decide which one we like better.

This issue is low-priority for us though, as the Vector3 seems workable enough to get us by in the initial test stages.

connor-wool, 2017-10-19:
Ok, once I looked at how easy it is to do this (publish as a point type) I added it to the code really quick. Closing issue.

connor-wool, 2017-10-19:
pull request is hung on linter issue, will need some attention still.

skallaher, 2017-10-19:
Please, in future, do not close issues until the PR solving it has been merged.

hmmwhatsthisdo, 2017-10-19:
Related: GH will close issues automatically if you put specific verbiage referencing the issue in a PR description or commit message associated with a PR. [Documentation here.](https://help.github.com/articles/closing-issues-using-keywords/)

skallaher, 2017-10-31:
@connor-wool, has this been fixed?

hmmwhatsthisdo, 2017-11-04:
This issue was resolved with #320.

closed 2017-11-04 by hmmwhatsthisdo