PalouseRobosub/robosub-2018 #317: Finalize Localization Methods

skallaher, 2017-10-17:
## Background
For the past few weeks, the Curiosity team has been tasked with analyzing the previous localization engine as well as a few alternatives.

## Action Required
- [ ] Decide on a method to localize the sub
- [ ] Analyze the amount of work the old system needs and provide a report on goals in the form of issues.

connor-wool, 2017-10-19:
From my subjective testing, I haven't been able to make any super great improvements in the accuracy of either the Kalman filter or the particle filter. I'd recommend actually building the testing system first, and then tuning the existing code. In addition, we need to get this code running on the sub as soon as possible in a pool test, and let it try to navigate along some points. We have no idea how this is going to run "real world"

connor-wool, 2018-01-30:
@skallaher I think this guy is ready to be closed.

closed 2018-01-30 by skallaher