PalouseRobosub/robosub-2018 #324: Conflicting TF coordinates published for cobalt

hmmwhatsthisdo, 2017-10-23:
When running Gazebo, the RViz bridge, and the Localization engine, the TF-published coordinates of `cobalt` (relative to `world` jump between the Localization engine's output and the origin point.)

```sh
roslaunch robosub_simulator gazebo.launch &
roslaunch robosub rviz.launch &
roslaunch robosub localization.launch &
rqt &
```
Inside of `rqt`, open a Plot window and add the following topics to the view:
* `/rviz/cobalt/pose/pose/position`
* `/real/position`
* `/tf/transforms[0]/transform/translation` 

Instead of matching `/rviz/cobalt/pose/pose/position` (the output from the localization engine, in this case), `/tf/transforms[0]/transform/translation` is oscillating between that and zero.

[image]

hmmwhatsthisdo, 2017-10-23:
Trial-and-error suggests the `sub_tf_broadcaster` node is publishing the incorrect TF data. Looking at [sub_tf_broadcaster.py](https://github.com/PalouseRobosub/robosub/blob/5b12824860db82ec148dc19e888470c2644f6af1/src/sensors/sub_tf_broadcaster.py) and the rviz launchfile seems to suggest it is indeed setting `cobalt`'s location to be 0 relative to `world`.

At this point, I'm unsure if that broadcaster is still needed, or if we want to simply augment its functionality. (@irwineffect may have more knowledge on its original/current purpose) Some ideas that come to mind are:
- Publish updated/zeroed transformation data for `cobalt` if no other transformation data is received for _n_ seconds.
- The above, but with the position of the `cobalt` model according to the `simulator_bridge` (assuming the bridge node is active)

irwineffect, 2017-10-25:
I think [sub_tf_broadcaster.py](https://github.com/PalouseRobosub/robosub/blob/5b12824860db82ec148dc19e888470c2644f6af1/src/sensors/sub_tf_broadcaster.py) should take in the position as reported by the localization engine and use that as the tf data for the `cobalt` frame. I think it would be a good idea to have a completely different tf frame for the actual position as reported by the simulator, perhaps `cobalt_sim` frame? Once we have that, we can have rviz display the sub where the localization says it is, and then have a "ghosted" sub displaying the simulator's reported position.

skallaher, 2017-10-25:
I like the way that @irwineffect suggests to do this. If it doesn't receive a point from localization, we can default to (0, 0, 0) or some other reasonable default.

skallaher, 2017-10-31:
@hmmwhatsthisdo, is this fixed by #327?

hmmwhatsthisdo, 2017-10-31:
Yes, this is - I thought merging #327 would have closed this, but apparently I forgot to reference it so GH didn't close automatically.

closed 2017-10-31 by hmmwhatsthisdo