PalouseRobosub/robosub-2018 #329: Yaw wrap-around causes velocity spikes

irwineffect, 2017-10-31:
When the control system calculates its orientation velocity, we have an issue where if the yaw wraps around (e.g. -180 to 180), we get a spike in our velocity. Below is an image showing it occuring in simulation:
[image]

This will cause problems if we use the derivative term for our yaw PID.

daymare, 2017-11-01:
We can wrap around the previous value and take the smallest distance from both the previous value and the wrapped value.

```c
int wrap(int value)
{
    if (value > 0)
        return value - 360
    else
        return value + 360
}
```

ryan-summers, 2018-01-03:
So I did some testing and this actually isn't an error for the D-term. Remember that we aren't taking the derivative of the yaw to get yaw velocity in the control system, but actually the derivative of the yaw's _error_. The error value is properly wrapped, so we don't see these spikes. I've attached a demo of the yaw wrapping with the proper derivative shown (note that this is after I have implemented a fix for derivative calculations as mentioned in https://github.com/PalouseRobosub/robosub/issues/330). 
[image]

closed 2018-01-03 by ryan-summers