PalouseRobosub/robosub-2018 #330: Control System Derivative Calculations

irwineffect, 2017-11-01:
When the control system runs at a significantly faster rate than our orientation sensors, it results in the control system intermittently spitting out zeros for the velocity/derivative, because the calculations run on values that haven't been updated since the last calculation. The control system should be able to gracefully handle this situation by recognizing when the orientation data hasn't changed since the last control update, and use the previous velocity as the current velocity.

btmoore, 2017-11-02:
There are a number of ways you can fix this, but generally speaking your control system shouldn't be running faster than your orientation sensors anyway. (I assume orientation sensors are your fastest sensors?)  It should be running at the same speed, or more properly, it should only be updating once your state (pos/ori) changes, whether that is at regular or irregular intervals.

If the state hasn't changed at all, then the calculations would just result in outputting the same control instruction anyway, so the updates would be unnecessary.  Unless you're thinking that you could extrapolate-forward what the position/orientation should be a fraction of a second later based on the velocity.  But the Control System is no place to do that - that's the localization engine's job.

I can think of a few reasons why you might do it differently, but in general, your control system update shouldn't be on it's own schedule, but rather be triggered after a localization update.

ryan-summers, 2018-01-29:
Closed due to https://github.com/PalouseRobosub/robosub/pull/349

closed 2018-01-29 by ryan-summers