PalouseRobosub/robosub-2018 #360: Control System Test Needs Updating

skallaher, 2018-01-27:
The current control system tests seem to be written for specific parameters. This should probably be remedied.

hmmwhatsthisdo, 2018-02-11:
In a similar vein, it appears that the current time-out of 120 seconds may be too short to guarantee the control system test passes - especially because the time-out is based on real time and not the simulator's (dilated) time. The VM on my 810 is currently unable to run tests as a result unless I push the time-out to 180 seconds. (I'm not 100% sure of the reason behind this, but it's possibly a result of the Spectre/Meltdown kernel patches combined with having to do software GPU rendering - I'm noticing a significantly higher proportion of kernel CPU time in `htop` than I did previously)

I'll most likely open a PR to change the time-out to 3 minutes instead of 2 - there's no particular downside I'm aware of to increasing the time-out.

irwineffect, 2018-02-11:
@skallaher, are you referring to the issue @hmmwhatsthisdo is talking about or something else?

skallaher, 2018-02-12:
@irwineffect, I was referencing a conversation I had with either you or @ryan-summers about the control system test being optimized for a specific set of PID parameters. I can't remember enough of the details at the moment.

ryan-summers, 2018-02-19:
Can we please clarify this issue to say that the overshoot and standard deviations allowed in the control system unit test need to be relaxed? I don't think that has been explicity stated anywhere here.

irwineffect, 2018-02-19:
Oh...... we just need to set more sane requirements for the control system test.

ryan-summers, 2018-02-19:
https://github.com/PalouseRobosub/robosub/pull/397