PalouseRobosub/robosub-2018 #376: Update kalman filter to read control system thruster messages

connor-wool, 2018-02-01:


connor-wool, 2018-02-28:
The current acceleration values for Cobalt are already being calculated in the control system as `m_accel`. We can add this Vector6 to the `control_status` topic, which we can then read from the Kalman Filter. This provides us with a pretty good estimate of linear acceleration to stuff into the Kalman Filter. Hopefully, this will replace our dependence (or at least reduce it) on the BNO and TRAX systems.

connor-wool, 2018-02-28:
Relevant files: 

`m_accel` is calculated on line 590 of file `control_system.cpp` but it doesn't look like the lifetime of the variable is just for the function call.

`getControlStatus()` (line 708) is responsible for creating the `robosub_msgs::control_status` message that is published on the `control_status` topic. 

I'm still trying to figure out where the `control_status.msg` file is hiding

skallaher, 2018-02-28:
@connor-wool, look at [control_status.msg](https://github.com/PalouseRobosub/robosub_msgs/blob/master/msg/control_status.msg)

connor-wool, 2018-02-28:
@skallaher Haha, that explains why I couldn't find them in the robosub repo. I thought I was going crazy.

btmoore, 2018-03-01:
Just a fair warning: the forces from the thrusters as information can be useful in your Kalman filter - but you shouldn't rely on those values for acceleration.  The control system works because it's set-up is robust to the actual system values.  Fundamentally, it decides it wants to move or spin in a direction, and increases power in that direction until it's satisfied.  Correspondence to real values are kind of ancillary.

In space that'd work fine, but because drag is a significant thing, within seconds of setting the thrusters to a particular thrust your net acceleration from them drops back to 0 as drag counteracts the motion.

I don't know hot much you guys are driving the current side thrusters, but if you're running them at about 1/3rd power and half thrust and topping out at about 0.5m/s,  that's 50N of forward thrust and reverse drag.  On a 50kg sub, that's an error of 1m/s^2 acceleration.  Not noise, but outright bias error. 

Since you're running a regular Kalman filter, and not an EKF or UKF, your drag model is going to be limited to being linear, rather than quadratic, so you're still going to be prone to significant error there.  To really do a proper drag model, you'd have to tack it on separately or somehow or another shoe-horn it in.

One way to do that would probably be a velocity look-up function.  Run the forward and strafing thrusters at varying power and see what the steady-state velocity is for the sub is at those thrust levels. Add in a time constant for accelerating between different thrust levels, and then integrate the velocity directly.  Then you can treat them as velocity measurements rather than acceleration measurements/kinematic models.  It'll still be a rough estimation, but it won't be nearly so likely to run away on you as trying to determine acceleration from the thrusters.

closed 2018-07-30 by connor-wool