PalouseRobosub/robosub-2018 #395: Calcultating linear acceleration from bno using vector calculus

trice9595, 2018-02-15:


trice9595, 2018-02-15:
We discussed this during our meeting on 2/12. I am unsure of the details of this. I added it to the backlog because it may be necessary and we should keep it in mind.

btmoore, 2018-02-16:
As a general notice, I have all of the math worked out for all of the different sensors.  Let me know if you guys get stuck on figuring anything out.

The matrix form for accelerometers is here under 'total matrix': http://robosub.eecs.wsu.edu/wiki/cs/localization/sensors/start#total_matrix

The accelerations felt by the accelerometer is a combination of four distinct phenomena.  The translational acceleration of the sub's center of mass, the rotational acceleration about that center, the centripetal acceleration from a constant rotational velocity, and the acceleration of gravity, which is dependent on the subs pitch and roll.   Some of these parameters depends on the position of the accelerometer relative to the sub's COM.  And of course, it assumes the accelerometer has the same orientation as the sub.

With only a single accelerometer, you'll need to determine rotational velocity and acceleration, and orientation, independently in order to calibrate out everything but the translational acceleration you're interested in.  The gyroscopes should tell you quite clearly the rotational components.  

The gravity component is messy because you're getting orientation from the same sensor, which is in large part using the accelerometer to determine its orientation in the first place.  However, you're better off using that than nothing.  You might be tempted to start off making the assumption the sub is always level, but gravity is a bit too strong for that. Even minor tilts will be significant.  9.8 x sin(5 degrees) = 0.86m/s^2 projected onto the x or y axes.  From a 5 degree tilt, after 1 second your dead-reckoning will predict you're 0.42m further than you ought to be.  And after 2 seconds, 1.6m.  

If you can't predict your pitch/roll within 1 or 2 degrees of accuracy to pull gravity out of your accelerometer measurement, you may be better off ignoring it.  You'd definitely have to assign a large measurement variance so that it doesn't assert itself too strongly over your other measurements or your kinematic model.

hmmwhatsthisdo, 2018-02-16:
@btmoore. I think the problem @trice9595 was referencing was that the Trax IMU does not, in its current state, provide the same sort of acceleration data (if at all) that the BNO nodes had exposed previously.

@ryan-summers, can you provide more details on what is/n't possible with the Trax IMUs? I thought you had mentioned previously that they can provide information about distortion of the gravity vector, but I'm not 100% certain this is the case.

ryan-summers, 2018-02-16:
All information available on how the TRAX operates and what is possible is available [here](https://www.tri-m.com/products/trim/spec-sheets/TRAX_User_Manual_r08.pdf). 

You will need to modify https://github.com/PalouseRobosub/robosub/blob/master/src/sensors/PniTrax.cpp and submit a pull request to add in the functionality you want. I would be happy to review it. Additionally, you can test code live on the submarine through the VPN - we keep the submarine powered on 24/7 now.